Polidea/RxBluetoothKit #304: readValue triggers onNext for observeValueUpdateAndSetNotification

Drusy, 2018-12-13:
**Describe the bug**

_Normal behavior_
For any characteristic that can notify, subscribe to `observeValueUpdateAndSetNotification`. `onNext` is called everytime the hardware notifies the characteristic.
Now at any time, call `readValue` on the same characteristic. As `readValue` is a `PrimitiveSequence`, the `onSuccess` callback is called with the updated characteristic value

I am using `CharacteristicIdentifier` for both notify and read subscriptions.

_Potential Issue_
Right before `onSuccess` is called on the `readValue` observable, I noticed that the `onNext` callback of `observeValueUpdateAndSetNotification` is also called. In my case, it results in a double notification and so a double management of the read value which could have unexpected behavior

**To Reproduce**
Steps to reproduce the behavior:
1. Subscribe to `observeValueUpdateAndSetNotification` and print any notified value on characteristic
2. Call `readValue` on this same characteristic
3. Observe the `observeValueUpdateAndSetNotification` being called

**Expected behavior**
The notification callback should not be triggered

**Environment:**
 - OS: iOS 9/10/11/12
 - Library version 5.1.3
 - Swift version 4.2

dariuszseweryn, 2019-01-11:
As far as I know this is the default behaviour of vanilla iOS API. There is a single callback for wether the value of a characteristic was updated by a notification or a read. I may be wrong but sorting it in the library may introduce some edge cases when a notification would be triggered at the time a read would happen. Thoughts?

dariuszseweryn, 2019-01-11:
We will get back to the topic next week probably