Polidea/RxBluetoothKit #309: discoverServices doesn't emit any result

vladiulianbogdan, 2019-01-30:
**Describe the bug**

The problem is that in our code we have multiple places where we discover services. The peripheral that the app connects to is also sometimes emitting a service change indication (I don't know if it is relevant).

What I've seen is that sometimes the `discoverServices` does not emit anything. I've deep dived into the code and the problem is around `remainingServicesDiscoveryRequest`. Sometimes the value in `remainingServicesDiscoveryRequest` remains unbalanced (it has value 1).

**Expected behavior**
`discoverServices` should always emit the discovered services.

**Environment:**
 - Device: iPod, MacBook Pro  2018
 - OS: 10.13.6
 - Library version 5.1.3
 - Swift version 4.2

c0diq, 2019-02-05:
Probably related to https://github.com/Polidea/RxBluetoothKit/issues/287

minixT, 2019-04-13:
@vladiulianbogdan can you confirm that this issue is fixed in version 5.2.0?

vladiulianbogdan, 2019-04-15:
Yes. I can't reproduce the problem anymore.

c0diq, 2019-05-16:
Can we close this now?

closed 2019-05-17 by minixT