Polidea/RxBluetoothKit #317: RxBluetoothService's scanningOutput won't output the repower-on devices

aelam, 2019-03-07:
**Describe the bug**
I can get the device from scanningOutput, then I power off the device and power it on again. I can't scan it out from the scanningOuput anymore.

**To Reproduce**
Steps to reproduce the behavior:
1. One the ExampleApp
2. Click on Start Scanning, then I can get my device on the list
3. Power off my device and then power on, I won't get any scan output of my device 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment:**
 - Device: [e.g. iPhone6, Macbook Pro 2018]
 - OS: [e.g. iOS8.1, macOS 10.11]
 - Library version [e.g. 5.1.2]
 - Swift version [e.g. 4.1]

Dong-Lin, 2019-05-15:
Add scanOptions will provide the fix for your problem
`                 let scanOptions = [CBCentralManagerScanOptionAllowDuplicatesKey : true]
                    self.centralManager.scanForPeripherals(withServices: nil, options:scanOptions)`

aelam, 2019-05-15:
@Dong-Lin  Thanks! I'll try

closed 2019-05-15 by aelam