Polidea/RxBluetoothKit #318: Different UUID from same BLE Device on iOS and Watchkit

rbrose, 2019-03-08:
**Describe the bug**
I have an iOS app and an Watch Extension. Both have RxBluetoothKit.

If i scan devices on iOS i get uuid 0A8CB164-1682-A7B7-9A97-F73D4C909731 for my device.
But if i scan on Apple Watch i get uuid B0ACF8CA-502F-F8DA-9F57-F2AC21ABE45F for the same device.

Why?

I scan devices on iOS and send uuid to apple Watch. Then the watch will connect to device and write value. But can't connect because not find with this uuid

How i can solve this problem? 

**Environment:**
 - Device: [e.g. iPhone5a, Apple Watch 4]
 - OS: [e.g. iOS12.1.4, watchOs 5.1.3]
 - Library version [5.1.4]
 - Swift version [4.2]

DvirTandemg, 2019-03-24:
Thats how apple works.
Your device generate an UUID that means that one phone generate UUID = 1 and another phone will generate UUID = 2 for the same BLE Device. 
It's per apple device and has nothing to do with the RxBluetoothKit.

dariuszseweryn, 2019-03-28:
Answer above is correct. Closing.

closed 2019-03-28 by dariuszseweryn