Polidea/RxBluetoothKit #321: Update library and example project to Swift 5

ptseng, 2019-03-26:
**Is your feature request related to a problem? Please describe.**
Desire to keep the library up to date.

**Describe the solution you'd like**
Update library and example project to Swift 5.

**Describe alternatives you've considered**
Doing it myself and submitting a pull request, as an alternative.

minixT, 2019-03-28:
Hi @ptseng , I plan to do this by the end of the next week but hope it will be sooner.

ptseng, 2019-04-08:
This PR looks like it's ready to go, can we please get a review?

https://github.com/Polidea/RxBluetoothKit/pull/322

randomiser, 2019-04-09:
I'm keen to see this change also... Getting dependency issues when I try and use RxBluetooth with RxSwift / RxCocoa etc.

nknotts, 2019-04-10:
It looks like this is merged, is there anyway to get a new release pushed to CocoaPods?

minixT, 2019-04-13:
Released version 5.2.0 with Swift 5 support

closed 2019-04-13 by minixT