Polidea/RxBluetoothKit #342: Multiple targets with same framework output name causes Xcode to build the wrong target

c0diq, 2019-06-01:
**Describe the bug**
Because of a bug with xcodebuild reported [here](https://openradar.appspot.com/20490378) and [here](https://github.com/Carthage/Carthage/issues/347), when RxBluetoothKit is integrated directly as a project (not Carthage) inside another project, Xcodebuild will incorrectly choose the wrong target (macOS, watchOS) when determining dependencies implicitly. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a simple project with just iOS as platform/sdk
2. Make sure Implicit Dependency is checked in the scheme
3. Add RxBluetoothKit as a dependency
4. Link RxBluetoothKit
5. Make sure it's not in Target Dependency 
6. On the command line execute:
`set -euo pipefail && xcodebuild -derivedDataPath 'build' -sdk 'iphonesimulator' -enableCodeCoverage YES  -scheme 'App' -configuration 'Debug' build`

Notice that it will randomly pick ios, watchos or macos.

**Expected behavior**
It should pick iOS. 

**Solution**
The way to solve it is to follow what other frameworks do like RxSwift, etc.. and have a single target `RxBluetoothKit` that supports all platforms.

**Environment:**
 - Macbook Pro 2018
 - OS: macOS 10.14.4
 - Library version: 5.2.1
 - Swift version: 5

m-ricca, 2021-11-11:
same problem

Marcello168, 2022-01-21:
same problem