Polidea/RxBluetoothKit #343: Cannot scan the peripherals

prijuly2000, 2019-06-16:
**Describe the bug**
Trying to scan the device which brodcasts serviceA
```
manager.observeState()
            .startWith(manager.state)
            .filter { $0 == .poweredOn }
            .timeout(3.0, scheduler: MainScheduler.instance)
            .take(1)
            .flatMap { _ in self.manager.scanForPeripherals(withServices: [BLEService.serviceA.uuid]) }
            .take(5.0, scheduler: MainScheduler.instance)
            .toArray()
            .subscribe(onNext: { peripheral in
                print("Scanned: \(peripheral)")
            })
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment:**
 - Device: [e.g. iPhone6, Macbook Pro 2018]
 - OS: [e.g. iOS8.1, macOS 10.11]
 - Library version [e.g. 5.1.2]
 - Swift version [e.g. 4.1]

prijuly2000, 2019-06-22:
@paweljaneczek @cwalcott anyone? Are you guys maintaining this lib? otherwise i will go with some other

cwalcott, 2019-06-23:
No, I'm not a maintainer. You could try posting on stackoverflow

dariuszseweryn, 2019-06-24:
Hi @prijuly2000 

Have you tried to debug the code? E.g. running without service filters and checking the results? Printing potential errors? Logging on particular steps of the flow? Or even turning the library logs on?