Polidea/RxBluetoothKit #357: Example code running iOS13 - Thread 2: signal SIGABRT Error

hohaidang, 2019-11-16:
iOS 13.2.2
[image]

robbaltzer, 2020-03-02:
Seems like this error started happening when I upgraded my apple devices to iOS 13.3.1. iOS releases before that seem to be OK.

robbaltzer, 2020-03-02:
Alamofire and other frameworks are having the same issue: https://github.com/Alamofire/Alamofire/issues/3051

Looks like it will be fixed in 13.4. Until then, this fixed it for me:

"
I had same issue with Alamofire, and solved it by removing use_frameworks! from pod file  then in console pod deintegrate, pod update and pod install
Now everything works fine.
"

closed 2020-04-03 by hohaidang