Polidea/RxBluetoothKit #362: Cannot connect to more than one peripheral?

EinatK, 2020-02-19:
Cannot call establish connection while connection has not been disposed. I understand not having multiple connections to one device, but what if I want to connect to more than one peripheral? 
CoreBluetooth allows connection to up to 8 peripherals. 
Do I need to create a manager for each connection?

Cierpliwy, 2020-02-19:
I should be possible to connect to different devices with the same BleManager. I need more information about your issue.

EinatK, 2020-02-19:
I have 2 different peripherals, with the same service that I need to be connected to simultaneously.
This is my connection code to the first one - I save it in a different place as a disposable which is not disposed. 
When i try to call establishConnection on a second peripheral, and assigning it to a second disposeable i get the error that i cannot call establish connection while connection has not been disposed. I need them both to be live connections, and to be able to dispose from them at separate times. 
When i set a second manager i manage to do so, but i want to do this with one manger, like CoreBluetooth allows  

`func getConnectedDevices() -> Observable<Peripheral> {
        return Observable.from(self.centralManager.retrieveConnectedPeripherals(withServices: [DeviceService.shipPrimary.uuid]))
            .toArray()
            .asObservable()
            .flatMap({ pers -> Observable<Peripheral> in
                print("array size: \(pers.count)")
                if pers.count > 0 {
                    let prevPer = UserDefaultsManager.getHaIdentifier()
                    for per in pers {
                        if per.identifier.uuidString == prevPer {
                            UserDefaultsManager.setOtherHaIdentifier(name: per.identifier.uuidString)
                        } else {
                            UserDefaultsManager.setCurrentLiveHaIdentifier(name: per.identifier.uuidString)
                            return Observable.just(per)
                        }
                    }
                    UserDefaultsManager.setCurrentLiveHaIdentifier(name: pers[0].identifier.uuidString)
                    UserDefaultsManager.setOtherHaIdentifier(name: pers[1].identifier.uuidString)
                    return Observable.from(pers)
                    
                    
                } else {
                    self.stateObserver.deviceStateSubject.accept(false)
                    return Observable.error(DeviceConfigError.deviceNotConnected)
                }
            })
            .asObservable()
            .flatMap({  per -> Observable<Peripheral> in
                print("PER CONNECTED? \(per.isConnected)")
                self.observePeripheralState(periphral: per)
                return per.establishConnection()
            })
    }`