Polidea/RxBluetoothKit #363: Move non-swift files out of `Source` folder

DevAndArtist, 2020-02-25:
Xcode 11.4 beta 2 reports an error which prevents our own package to compile which depends on `RxBluetoothKit` because `RxBluetoothKit` contains non-swift files in its `Source` folder. Please move those into a different place.

DevAndArtist, 2020-03-24:
Xcode 11.4 is out and this prevents projects from compiling.

SaintNicholas, 2020-03-25:
I'm having the same problem.

lfongaroScp, 2020-04-01:
Is the library not longer maintened?

akmarinov, 2020-04-26:
This PR should help - https://github.com/Polidea/RxBluetoothKit/pull/368

Unfortunately the project seems to be dead with no development in almost an year, so I'd look into migrating away from it instead.

lfongaroScp, 2020-04-27:
> This PR should help - #368
> 
> Unfortunately the project seems to be dead with no development in almost an year, so I'd look into migrating away from it instead.

I'm planning to do the same. Do you have any suggest?