Polidea/RxBluetoothKit #374: Connection disconnection notification not coming

SekhonGurpreet, 2020-07-13:
Hi,

I am using RXBluetoothKit to pair with BGM meter. First time I am able to get the scanned response and able to pair with the meter. After that When I switch off the meter, I am able to get the notification that meter is turned off. But after that if I again turn on the meter, I am not getting any response in the scanned response.

Thanks,
Gurpreet

dariuszseweryn, 2020-07-13:
The topic and the post you have written do not match.
> Connection disconnection notification not coming

vs
> After that When I switch off the meter, I am able to get the notification that meter is turned off.

So it seems that you get the disconnection callback?
> But after that if I again turn on the meter, I am not getting any response in the scanned response.

Are you still scanning at that time? Maybe you do not get more scan callbacks because they are duplicates (there is an option for that)? Btw. once you have device ID you can connect to it without scanning (unless it is not bonded and changes its MAC address).

SekhonGurpreet, 2020-07-13:
Yes, I am still scanning. I am able to see the other bluetooth meter in the scanned response. I am having the device id but I don't know when the user is switching the meter on/off, as I am not getting any callback.

dariuszseweryn, 2020-07-13:
You said that:
> After that When I switch off the meter, I am able to get the notification that meter is turned off.

This is contradictory. Could you elaborate _what_ do you do with the library exactly? Or even better simply [conform to the issue template](https://github.com/Polidea/RxBluetoothKit/blob/master/.github/ISSUE_TEMPLATE/bug_report.md) you have ignored?

SekhonGurpreet, 2020-07-13:
Yes, For the first time when I switch off the meter, I am getting callback. But after that not getting any callback whether I am switching the meter on/ off. 
One thing I forgot to mention, If the meter is having any pending glucose to sync or If I change  any setting in the meter itself then I am getting callback. In Nutshell, If there is any change in the bluetooth meter then I am able to get the callback.

Usage - We are scanning Blood glucose meter, Pairing/ Unpairing. We are getting information for those meters using various BLE services and characteristics.

dariuszseweryn, 2020-07-13:
> One thing I forgot to mention, If the meter is having any pending glucose to sync or If I change any setting in the meter itself then I am getting callback. In Nutshell, If there is any change in the bluetooth meter then I am able to get the callback.

This would mean that you need to consult the specification of your glucose meter. Not sure how the library can help you – if your peripheral does not advertise then it cannot be connected.

SekhonGurpreet, 2020-07-15:
I have verified my peripheral is advertising. I can see the same device in the nRF Toolbox app.

dariuszseweryn, 2020-07-15:
Ok. I lost you. Could you try to recap what exactly you have problem with? Showing BLE logs and code can be helpful.