Polidea/RxBluetoothKit #388: peripheral.canSendWriteWithoutResponse change to false after enter background

MrQQQQQQQQQQ, 2020-12-01:
peripheral.canSendWriteWithoutResponse change to false after enter background,then  the in the function : "writeValue(_ data: Data,
                           for characteristic: Characteristic,
                           type: CBCharacteristicWriteType,
                           canSendWriteWithoutResponseCheckEnabled: Bool = true) -> Single<Characteristic>".  will not send the data.

**To Reproduce**
Steps to reproduce the behavior:
1. scan and connect to a peripheral, and keep sending  datas 
2. click Home, Makes the App enter background
3. then will find  canSendWriteWithoutResponse property changed to false .
4. Even the App enter Foreground, the property  canSendWriteWithoutResponse will never change to true

**Expected behavior**
canSendWriteWithoutResponse will never change to false 

**Environment:**
 - Device: [iPhone7, Macbook Pro 2018]
 - OS: [ iOS13.5, macOS 10.15.7]
 - Library version [6.0.0]
 - Swift version [5.2]

dariuszseweryn, 2020-12-01:
Hello,
Are you sure that it is not a system restriction?

MrQQQQQQQQQQ, 2020-12-01:
> Hello,
> Are you sure that it is not a system restriction?

I am not sure if it is a system restriction. but if yes, why the property canSendWriteWithoutResponse won't change to true when i back to foreground. in my app, i used a BLE background mode ,so my app won't  be suspend by system.the app can still invoke the write method.but because of this property, it will never really invoke "writeOperationPerformingAndListeningObservable"