Polidea/RxBluetoothKit #396: DO NOT USE THIS LIBRARY!!!! - Reasons why below

xavierbr, 2021-04-27:
This library does not allow your app to run in background (iPhone sleep/screen off) and continue to send BT commands while in this state. This is like the most common use case for BT peripheral/app scenarios as users almost never leave their app and phone awake and in foreground. Additionally, it seems like Polidea has been acquired by another company and support for this library is likely non-existant as time goes by. If your app is mission critical, use CoreBluetooth or you're going to end up having to support this project on your own.

shalinipk, 2021-05-04:
@xavierbr as long as we specify "bluetooth-central" in info.plist -> UIBackgroundModes, shouldn't app run in background mode ? 

agree about the support, applies to pretty much all third party frameworks out there. Trade off between quick deployment vs long term maintenance, I guess

XabierGoros, 2021-07-08:
As @shalinipk said, I don't think this library has anything to do with the working modes of the host app. The library just enables you to use the CoreBluetooth API in a more reactive and easier way for us, nothing more. The same changes you have to do using CoreBluetooth to support background mode, may also be done using this library.

However, even though the owner of the library is not present any more, the library is still open sourced and we can collaborate to improve it. I guess if you find a better way to support those `UIBackgroundMode`, a PR will be more than welcome for the community. This is why I'd sincerely avoid creating issues on this way with that kind of title, it could be written or proposed in another way with a simple support/help approach.

fnazarios, 2022-01-07:
I have an app that runs in the background perfectly. Including Bluetooth restoration. 🤔

closed 2022-01-12 by xavierbr