Polidea/RxBluetoothKit #414: Compilation fails with Xcode 14 and Xcode 13.4.1, is this lib still being maintained?

AniOSDeveloper, 2022-09-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment:**
 - Device: [e.g. iPhone6, Macbook Pro 2018]
 - OS: [e.g. iOS8.1, macOS 10.11]
 - Library version [e.g. 5.1.2]
 - Swift version [e.g. 4.1]

Serg-Pogrebnyak, 2022-09-16:
maybe it not compile for iOS15 and higher?

bonzairios, 2022-11-24:
Doesnt look like it.  My project stalled since I updated to xcode 14, now I get a bunch of errors when building.  Not sure what to do now.

Serg-Pogrebnyak, 2022-11-24:
This library not updated to support iOS 15 and higher, try to make some changes manually in the library from [this](https://github.com/Airthings/RxBluetoothKit/pull/1/commits/b24a4ff84874123d2f0eeab88b1e39cf6237f5a0) merge request. Or you can try to find fork of this library with fix.