Polidea/RxBluetoothKit #420: Keep RxBluetoothKit Alive

fraune, 2023-07-05:
It appears that Polidea/RxBluetoothKit has not reviewed or merged any Pull Requests since 2020.

I would like to help keep this library alive. Could someone with the appropriate access help by doing one or more of the following?

* Become active in engaging with the community again
* Grant me permission to review/accept Pull Requests
* Transfer ownership of the library to a willing maintainer

I would love to hear thoughts and ideas around this, since there is not another Bluetooth library for RxSwift of this size and reliability.

_Cross-posted on the RxBluetoothKit Glitter._

_Tagging anyone that GitHub says is on the Polidea team_
@adam-stasiak @agatapst @jakublipinski @kkucharc @literator @potiuk @slawomir-andrian

potiuk, 2023-07-05:
Poliidea has been acquired and unless some individuals want to keep it running, I think best approach is to fork it and develop on your own. The Apache licence allows it, you can fork it and release/develop on your own. I think there is no need to transfer ownership.

fraune, 2023-07-05:
My concern is that there are already 300 forks of this repo, and I believe most are for adding a feature or fixing a bug that specifically affects an individual or single team. As time goes on, it will become more difficult to source great code from many individuals across many forks.

potiuk, 2023-07-05:
BTW. You can start with gathering those 300 repos and bringing their fixes into your fork and ask them to switch to yours. That's what I'd do if I wanted to provide value for others.

fraune, 2023-07-05:
I understand. I've never attempted something like that before, but I think it may be worth the effort. Thanks for your perspective and for letting me know that this repo probably won't receive any more updates.

jberkhei, 2024-09-18:
I'd be interested in helping keep this alive.

Serg-Pogrebnyak, 2024-10-02:
Some time ago I tried to do something the same, I found info that Polidea was redeemed by another company - Snowflake, and I wrote them and asked to move this repo to open source. I wrote an email to them, tried to make an account on their website, and submitted a technical request, but I had no luck. Maybe if someone contacts them via Linkedin or located in Poland and can call them and tell them about this problem - it would be perfect. But after one and half years no updates about this from my end...

uKL, 2024-10-24:
The Android counterpart has been moved to the maintainer private github. Maybe there is hope? @dariuszseweryn

dariuszseweryn, 2024-10-24:
I will check what could be done

dariuszseweryn, 2025-03-12:
@jberkhei or @Serg-Pogrebnyak is any of you still interested in taking ownership and maintaining this library?

Serg-Pogrebnyak, 2025-03-12:
@dariuszseweryn I've made a fork and supports my version, but if we can move it to the open source - that will be great! Do you have any news about that?

dariuszseweryn, 2025-03-12:
Yeah, it took a while but I got the necessary approvals. If you would like to take the ownership of the repo going forward it is possible to transfer it to your GitHub account. (That being said I am unsure how this will behave if you have a Fork already but maybe it would just merge?)

I would need your legal name and email address so appropriate documents could be sent.

jberkhei, 2025-03-12:
I'd be happy to help as well, I am using this library via my fork on numerous projects.  Not sure I want\need ownership.

Serg-Pogrebnyak, 2025-03-12:
@dariuszseweryn Thank you so much for your trust, but I would prefer for this library to be transferred to the community and maintained collectively.

Mr. @DaBs previously [mentioned](https://github.com/Polidea/RxBluetoothKit/issues/398#issuecomment-915204232) a proposal about moving this repository to the community organization. Mr. @DaBs could you help with this process?

@dariuszseweryn I truly appreciate all the work you’ve put into this and the fact that this issue hasn’t been forgotten. Thanks again!