Polidea/RxBluetoothKit #422: Cannot specialize non-generic type 'Observable<Element>'

avinashamanjha251, 2023-09-27:
Hi Dev team,
I just install RXBluetoothkit with Xcode 15 and start getting "Cannot specialize non-generic type 'Observable<Element>'".

[image]

Thanks in advance.

jberkhei, 2023-09-29:
You can get it to compile by updating it to this, however you will then run into RXSwift 6 issues.

return Observable.empty()

torhenningegenes, 2023-10-03:
I am having the same problem. Would love to hear i anyone has gotten around this including the RXSwift 6 issues.

jberkhei, 2023-10-03:
People have forked this and fixed it.  My work arounds.

pod 'RxSwift', '~> 6.0'
pod 'RxCocoa', '~> 6.0'
pod 'RxBluetoothKit', :git => 'https://github.com/jberkhei/RxBluetoothKit.git', :branch => 'master'
pod 'RxMKMapView', :git => 'https://github.com/jberkhei/RxMKMapView.git', :branch => 'master'
pod 'RxRealm', :git => 'https://github.com/jberkhei/RxRealm.git', :branch => 'main'

torhenningegenes, 2023-10-03:
Thank you for answering so quickly. Didn't run into any issues with RXSwift when I removed the `Element` type. At least not yet. 🙂

avinashamanjha251, 2023-10-03:
Here's my side update. 

[image]

closed 2023-10-03 by avinashamanjha251